fieldOfView/Cura-ArcWelderPlugin #2: The plugin launch the conversion also when we save the project

5axes, 2020-11-28:
`        gcode_dict = getattr(scene, "gcode_dict", {})
        if not gcode_dict: # this also checks for an empty dict
            Logger.log("w", "Scene has no gcode to process")
            return`

It prevent to launch the conversion if no Gcode have been calculated .  But not if we save the project . Any solution to avoid this case ?

fieldOfView, 2020-11-30:
This is true for all postprocessing-type plugins. It may be hard to fix this from inside the plugin, but I will keep looking for a way to detect which filewriter is being used by the outputdevice.

FormerLurker, 2020-12-14:
Anything I can do to help?

fieldOfView, 2020-12-14:
Not unless you want to change how Cura works internally ;-)

Like I said, this is true for all postprocessing-type plugins, including the postprocessor that ships with Cura. It needs to be fixed in Cura, not in this plugin.