fieldOfView/Cura-ArcWelderPlugin #6: help with using

dacentichka, 2021-01-14:
I downloaded this plugin from the ultimaker store, restarted the slicer, loaded the model and clicked generate code, after which I saved it, for verification I opened it with pad ++ notes, and used G2 &G3 pointers, and could not find these commands in the code. please explain why?

5axes, 2021-01-16:
Did you activate in the special mode section the Arc Welder option ?
[image]

fieldOfView, 2021-01-16:
What OS do you use? There's probably an issue with the OS X version of the ArcWelder executable that does the heavy lifting.

fieldOfView, 2021-01-16:
Could you try this [development snapshot](https://storage.googleapis.com/ultimaker-cura-packages/v1/packages/ArcWelderPlugin/zzHJfdU_udY4lWeMO9r9Ha-zSXKtZbLMRyRejKldEXtt/ArcWelderPlugin-v7.0.0-2021-01-16T14%3A43%3A57Z.curapackage?Expires=1926168243&GoogleAccessId=cura-packages-api-production%40stardust-193112.iam.gserviceaccount.com&Signature=iHzmKmGOpJJ%2F%2BD0fpMTWNgIhLgKY%2FRNkItlpuKHF8lUPCMkf2XZfQ7G7IIWSpuCcfmgbFNbBZprub9Gg9UZIkbTNML%2FZ8PGNPp18Bt7CzVoDEzAG2lln1AAGHcHbCQYPB4ZYsBSGcmKBr24XtLyWs%2BnatOkrkqaZzpKzhuxTFS2SZ8cyN%2BUfAqVOraeOSNbUULZe5Y0Oanb5oTyJXJxWR8GzC1wXPAwKZmV3GTb%2F0EgWLhd4IiSX6Rn%2BpKSbMVm%2Be3WLRSayrh0Gz%2FuBQujQnwpvvMic95wOvOjOIKZE9h9D9RhpngXLKYR6BSoymCFGQFoOVn%2F6sTElYOsnrZw9HQ%3D%3D)? Download the file and drop it onto the buildplate in Cura, as if you were opening a 3d model.

This development snapshot contains a different version of the ArcWelder executable for OSX.

cc @3dprintedmac

3dprintedmac, 2021-01-16:
Checked gcode after dropping in the new plugin and restarting and not seeing any change. I'm currently using the "save to file" and "print to octoprint" buttons to save / send the gcode. Is that correct?  I'm running Windows 10 with my Cura. Tried with Cura 4.7.1 and now also with 4.8. Also, am i supposed to have settings i can see in the extension window because no settings show up? 
[image] 

Thanks for the help!

fieldOfView, 2021-01-16:
Ah, your github username confused me. I thought for sure you are an OS X user.

The plugin does not add anything to the Extensions menu, but it add settings as shown by @5axes above. Do you see those settings?

3dprintedmac, 2021-01-16:
hmm, i am not seeing those settings anywhere. It would be in the setting menu along side speed, support, infill, etc right? It does show it installed in the plugin menu, i keep checking that ha. 
And yeah, i use mac for most other things but not my printing really!

fieldOfView, 2021-01-16:
Above the Custom settings list, there is a Search settings... field. Do you get any settings if you type "Arc" there?
If not, could you upload and post a link to your cura.log? It can be found via Help -> Show configuration folder.

3dprintedmac, 2021-01-16:
I do not get any arc settings when i search it there. 
[cura.log](https://github.com/fieldOfView/Cura-ArcWelderPlugin/files/5824827/cura.log)

fieldOfView, 2021-01-16:
Thanks for the log. It looks like there may be a dependency (dll) for the ArcWelder executable missing in your system.

Could you download the application "Dependencies" from https://lucasg.github.io/Dependencies/, extract that zip and run `DependenciesGui.exe`?

In the Cura configuration folder, go to `plugins\ArcWelderPlugin\ArcWelderPlugin\bin\win64` and drag and drop ArcWelder.exe into the DependencesGui window. Hopefully that will tell us what is missing.

3dprintedmac, 2021-01-16:
[image]

Does this mean i'm missing that "VCRUNTIME140_1.dll"?

seantapscott, 2021-01-17:
As a side note, I had no idea there was a special "Arc Welder" option in the Special Modes menu.   May want to add this to the ReadMe, because it is not obvious.  

On a second side note, this just compressed a gcode file from 28MB to 8MB, which is incredible. Almost all the shapes were just a series of tiny arcs, so super ideal. 

Thanks

fieldOfView, 2021-01-17:
> Does this mean i'm missing that "VCRUNTIME140_1.dll"?

It looks like it. Here is another [development snapshot](https://storage.googleapis.com/ultimaker-cura-packages/v1/packages/ArcWelderPlugin/zzHJfdU_udY4lWeMO9r9Ha-zSXKtZbLMRyRejKldEXtt/ArcWelderPlugin-v7.0.0-2021-01-17T13%3A12%3A21Z.curapackage?Expires=1926249145&GoogleAccessId=cura-packages-api-production%40stardust-193112.iam.gserviceaccount.com&Signature=Pb45YhI3vFM1uYSg%2FoA5TbUwfx86Tulw%2BD9kTdp5NeFxvVXkHHSUhKClmO8f8Ss5qMXXu6y3ddz7xySnr%2FZ1sg%2FvnQ0K3Mtp59tnb%2FUXUfn4XXeeEtiYxi%2BUgz0ssODaR5%2B14BhDK2pQ0Bp2sPzbI%2BtST69XRLEr8dSNomXczTpyFzWz4bN33W1J5OeQTVDlNLGmjSNbZO4e6RRMk7sMl%2BsrGK9UZ3cz3HMsx%2B4Buu8y7c0EejrDECrQPsCNZZZw5h8%2FXshXzw1UYvL5KSNHtTfTrgt1CqSM6UoJpZY3li8YpREjIsfcO73DrfdBA5r6%2BuxVe99R9spUVA7UkoF2%2Bw%3D%3D), which includes that dependency so hopefully you don't have to install it yourself.

fieldOfView, 2021-01-17:
>  May want to add this to the ReadMe, because it is not obvious.

I would agree, but that ReadMe is pretty much inaccessible from inside Cura, so I will see if I can add a way to show the readme in Cura, and/or add this tidbit of information in the Marketplace description of the plugin.

5axes, 2021-01-17:
> > May want to add this to the ReadMe, because it is not obvious.
> 
> I would agree, but that ReadMe is pretty much inaccessible from inside Cura, so I will see if I can add a way to show the readme in Cura, and/or add this tidbit of information in the Marketplace description of the plugin.

On my point of view the Arc Welder option should be by default activated . It could suppress some beginner "Issue" and it's not illogical , if you install this plugin , the first aim is to test the result on your printer.

seantapscott, 2021-01-17:
I think both avenues are effective. Thank you again.

dacentichka, 2021-01-18:
[image]
[image]
good afternoon, I tried to reinstall the plugin by dragging the file from the cometaries above into the kura window, in the plugins folder it appeared but in the special modes field it did not become available, I tried to install a couple more files from the rep rap branch and there is also nothing there.

fieldOfView, 2021-01-18:
> I tried to reinstall the plugin by dragging the file from the cometaries above 

Which one? Try the one from [this comment](https://github.com/fieldOfView/Cura-ArcWelderPlugin/issues/6#issuecomment-761810702)

>  I tried to install a couple more files from the rep rap branch 

I have no idea what files from what "rep rap branch" you are talking about.

dacentichka, 2021-01-18:
я пробовал сегодня уже то что вы мне прислали и пробовал из статьи по этой ссылке https://community.ultimaker.com/topic/34888-arc-welder-anti-stutter-plugin/

dacentichka, 2021-01-18:
from the article on the link that I sent, only one item appeared in the settings, and it did not work well, there were a lot of printing defects

dacentichka, 2021-01-24:
good afternoon, I have solved my problem, now I will tell you what I tried and what helped me. I have Windows 10, I tried to install from the store in the kura itself, the plugin folder did not even appear in the Kura plugins folder, then I dragged the downloaded file from the comments above, and also nothing changed, then I renamed the downloaded file .zip unpacked it and take it from there vail .exe for Windows, and also nothing has changed, then I just took everything that I unpacked from this file and put it in the kura plugins folder and everything worked, while if you go to the plugins section in kura, she said that everything is installed, before putting everything in the plugins folder, I removed everything from the applications section, then put everything in the plugins folder, restarted kura, and lo and behold, everything worked, tonight I will test.

fieldOfView, 2021-01-25:
Here's a new [development snapshot](https://storage.googleapis.com/ultimaker-cura-packages/v1/packages/ArcWelderPlugin/zzHJfdU_udY4lWeMO9r9Ha-zSXKtZbLMRyRejKldEXtt/ArcWelderPlugin-v7.0.0-2021-01-25T11%3A08%3A03Z.curapackage?Expires=1926932894&GoogleAccessId=cura-packages-api-production%40stardust-193112.iam.gserviceaccount.com&Signature=N8jg1rq3JW0q%2BeRHxmIRNaq5Lb87wH0yEoUrNUB6jLMAkVWlnKGGSHsnnqWhVMa7VYikI6vDzdxju3JGl0QS4CUI8h3HCRoWdlrmaYBb6oU9rm2E8pknuTxHWEmNdTjEyAJ3FhKANYi2%2BoYE8NsoxAwA%2FLmXEKdr%2FGE8smqWV4qgWhJNv1eITT%2F8pKChkZR3MmFIF%2FzjHHjSLWks4UjLeX03IUYTrozi3tpv2E80ebfXDzEtQMvTALsVjoDSOByy7xWNId3zAC2iZAdxplzyW54z8pW3xOYLae3lM0cHLhcfbOJMBGLctntAehlXes0DmhLvEjptwbPCRvBylaHCGg%3D%3D) to try, including the latest version of the ArcWelder executables. 

For more information, see the ArcWelderLib 1.1 release notes:
https://github.com/FormerLurker/ArcWelderLib/releases/tag/1.1.0

tablatronix, 2021-05-10:
There are 2 places saying optional or toggle would be useful or add some kind of instruciton that this is a setting..
[image]
[image]

closed 2021-07-20 by fieldOfView