fieldOfView/Cura-ArcWelderPlugin #7: Plugin installed but no evidence of change

3dprintedmac, 2021-01-15:
Just installed this plugin hoping to do the conversion from the cura side instead of octoprint so i can still print directly from cura to octoprint. I installed the plugin but not seeing any evidence of it. Should there be a menu option in Extensions? Should it be automatically converting my file when i send it? When i look at the gcode, it doesn't seem to have any G2 or g3 commands added, so doesn't appear to have changed anything.

Maybe i'm missing instructions somewhere on how to have it work, so any help or tips would be fantastic, thanks!

5axes, 2021-01-16:
Did you activate in the special mode section the Arc Welder option ?
[image]

fieldOfView, 2021-01-16:
I'm closing this as a duplicate of #6. Please continue there.

closed 2021-01-16 by fieldOfView