fieldOfView/Cura-ArcWelderPlugin #8: Gcode results mutilate file

seantapscott, 2021-01-17:
So the resultant post-processed file appears to mutilate my models.  What were supposed to be simple semi-circle arcs of 2mm thickness with filleted edges has these strange lines across the semi-circles' openings. 

[image]

[WheelHolders.zip](https://github.com/fieldOfView/Cura-ArcWelderPlugin/files/5826561/WheelHolders.zip)

The slicer is Cura 4.8.0 for an Ultimaker 3E printer.  Slicer is set up for Griffin gcode flavor.  Arc Welder is enabled.
[UM3E_WheelHolders_43g_arcs.zip](https://github.com/fieldOfView/Cura-ArcWelderPlugin/files/5826562/UM3E_WheelHolders_43g_arcs.zip)

seantapscott, 2021-01-18:
I realize that this project is just a wrapper for the original project, so I'm ok with just closing this if you like, and I'll open a case upstream. 

Sean

fieldOfView, 2021-01-18:
To be honest, I don't expect much improvement in print quality on Ultimaker printers in the first place. G2/G3 support in the Ultimaker printers is vestigial at best. Also Cura does not display G2/G3 movements correctly. I'll try to improve the latter in future versions of the plugin.

seantapscott, 2021-01-18:
I know Cura doesn't preview the arc code properly, but what it printed did have a strange line across the front of the semi-circle. Like I said, I tried reading through the gcode for ifgure out why, and I'm not really sure.  Just sharing an example if it might help at all. 

Thank  you again, love your support and investment.  Making the world a brighter place.

fieldOfView, 2021-01-19:
I think it is best if you shared your result in the ArcWelderLib repository. I don't know enough about the inner workings of ArcWelder to say what is going on/wrong.

seantapscott, 2021-01-19:
Here is my post.  I'm going to go ahead and guess this is a UM3 arc interpretation problem.  I was able to install Simplify3D and get a preview of the gcode, and it look accurate, though there was a skirt running through it without the proper offset applied in the preview. 
[image]

fieldOfView, 2021-07-22:
Since the linked ArcWelderLib issue is closed, I'm closing this one too.

closed 2021-07-22 by fieldOfView