fieldOfView/Cura-ArcWelderPlugin #10: Plugin not available on Cura 4.8, because utf8 problem comes up

unlimitopen, 2021-02-06:
Hallo together,

if you install the plugin and after restart of cura, following problem in the cura log appears and the plugin not available on the experimental options.

`2021-02-06 10:59:14,677 - ERROR - [MainThread] UM.Logger.logException [106]: Exception: Error loading plugin ArcWelderPlugin:
2021-02-06 10:59:14,689 - ERROR - [MainThread] UM.Logger.logException [110]: Traceback (most recent call last):
2021-02-06 10:59:14,694 - ERROR - [MainThread] UM.Logger.logException [110]: File "X:\4.8-exe\build\inst\lib\python3.5\site-packages\UM\PluginRegistry.py", line 480, in loadPlugin
2021-02-06 10:59:14,698 - ERROR - [MainThread] UM.Logger.logException [110]: File "C:\Users\Unlimitopen\AppData\Roaming\cura\4.8\plugins\ArcWelderPlugin\ArcWelderPlugin\__init__.py", line 11, in register
2021-02-06 10:59:14,702 - ERROR - [MainThread] UM.Logger.logException [110]: return {"extension": ArcWelderPlugin.ArcWelderPlugin()}
2021-02-06 10:59:14,705 - ERROR - [MainThread] UM.Logger.logException [110]: File "C:\Users\Unlimitopen\AppData\Roaming\cura\4.8\plugins\ArcWelderPlugin\ArcWelderPlugin\ArcWelderPlugin.py", line 51, in __init__
2021-02-06 10:59:14,708 - ERROR - [MainThread] UM.Logger.logException [110]: version_output = subprocess.check_output([self._arcwelder_path, "--version"]).decode()
2021-02-06 10:59:14,712 - ERROR - [MainThread] UM.Logger.logException [110]: UnicodeDecodeError: 'utf-8' codec can't decode byte 0xf6 in position 22: invalid start byte`
I have in the original username of the file path a german umlaut, is thats the problem, why the plugin not appears in the "experimental" options on cura?

fieldOfView, 2021-02-06:
>  is that the problem, why the plugin not appears in the "experimental" options in cura?

It appears so.

fieldOfView, 2021-02-06:
Should be fixed with the above commit. I'll keep this issue open until a fixed version is available in the Marketplace.

fieldOfView, 2021-02-07:
3.5.1 is available in the Marketplace.

closed 2021-02-07 by fieldOfView

unlimitopen, 2021-02-07:
Thanks a lot, it works perfectly, thanks for the very fast response and correction!!

commit de11907e (2021-02-06): Fix running ArcWelder with non-ascii characters in full path

Fixes #10