fieldOfView/Cura-ArcWelderPlugin #11: Request: Better description as to where the option is turned on/off

RFBomb, 2021-02-08:
Since this is by nature a post-processing script, I assumed it would be available in the post-processing script section of CURA. 
The only way I found out that it was inside of 'Special Modes' was by coming here and seeing someone else's issue about it not showing up in their 'special modes' section (issue was already closed). 

Please add a sentence or two to the main description on here and on the CURA Marketplace that tells users that it adds the toggle under the 'special modes' section in their settings menu. (If it already exists, then it wasn't apparent enough, because I missed it...)

Thanks for creating this wrapper. Worked well.

b-morgan, 2021-06-29:
Please add some documentation on how to use this plugin.

fieldOfView, 2021-07-16:
Added documentation here: b15b56bacff9388ab901c3c8cc2b622a20e22d15

closed 2021-07-20 by fieldOfView