fieldOfView/Cura-ArcWelderPlugin #12: hard parameter limits prevent from using sensible values

wohltat, 2021-02-24:
if the chosen values for the cura arcwelder plugin parameters exceed the minimum or maximum values, then the plugin refuses to process completely. This is no good behaviour. Especially because it prevents from using sensible values for parameters like
```
    "arcwelder_resolution":
    {
        "minimum_value": 0.025,
        "maximum_value": 0.075,
```
There are definitely usable values outside that range. 

A better way would be make soft limits that give a warning but to not stop the processing.
Alternatively just drop the limits altogether and put the hints in the parameter description.

fieldOfView, 2021-02-25:
You are quite right. I think the minimum and maximum were introduced as a misinterpretation of the default value mentioned here:
https://github.com/FormerLurker/ArcWelderLib#resolution-maximum-path-deviation

'Default: 0.05 (+- 0.025)' was interpreted as the value for the setting being limited. It would seem that is incorrect. I'll remove the artificial limits.

fieldOfView, 2021-02-25:
Fixed in 826d568, but I'll leave this issue open until a version containing this fix is released in the Marketplace.

wohltat, 2021-05-17:
is it not released in the marketplace yet?

commit 826d568f (2021-02-25): Remove incorrect minimum/maximum values for resolution

Fixes #12
closed 2021-07-20 by fieldOfView