fieldOfView/Cura-ArcWelderPlugin #13: Request for version 1.1.1 in Cura Marketplace

seantapscott, 2021-02-25:
There's a bugfix version of the ArcWelder plugin (v1.1.1) available [here](https://github.com/FormerLurker/ArcWelderLib/releases/tag/1.1.1) . 

My current version seems to still be using 1.1.0 (discovered using arcwelder.exe -h ).  

Would you be willing to push a new version into the cura marketplace?  

Please and thanks. :)

fieldOfView, 2021-02-25:
The next version of the plugin will include ArcWelder 1.1.1 or newer. I have been told by @FormerLurker that the 1.1.1 update was probably not all that relevant for Cura, which is why I have not yet put much effort into that.

fieldOfView, 2021-02-25:
I have updated the binaries in commit 2c9ebb, but will keep this issue open until a new release is available in the Marketplace

seantapscott, 2021-02-25:
Thank you.  I have been reading some of the issues that users are having,
and some of them do seem relevant.

Thanks again.

FormerLurker, 2021-02-25:
@seantapscott, one fairly major bug was fixed in 1.1.1, but it seemed pretty rare.  It was related to whitespace between G1s that were converted into arcs.  It was a pretty rare situation.

The only issue I can't seem to track down is why some users have slowdowns when printing infill.  I believe that might be a firmware problem, but I've received enough gcode samples now to figure this out.  I'm suspecting that some firmware doesn't properly handle feedrate changes done with G2/G3, but will need to confirm that.

Thanks @fieldOfView for keeping things updated!  The algorithm is starting to mature, and I expect the release frequency to decrease.

closed 2021-07-20 by fieldOfView