fieldOfView/Cura-ArcWelderPlugin #14: No skirt or walls with arcwelder enabled

pbezza, 2021-03-14:
I enabled arcwelder and sliced. I had 3 walls, 3 bottom layers and a skirt in my settings. The printer skips the skirt, prints the bottom 3 layers with no walls and then prints the infill still without any walls.

I disabled arcwelder and sliced again and it prints correctly.

seantapscott, 2021-03-14:
Hi Paul,

I help out here sometimes by asking verifying questions and gathering
information for troubleshooting.  Could you share your firmware version,
printer model, and slicer type?

pbezza, 2021-03-14:
AnetA8 with MKS GEN L V1.0 and Marlin bugfix v1.1.9. Using Cura v4.8.0.

seantapscott, 2021-03-14:
ArcWelder requires support for G2 and G3 arc commands. This is possibly a
case of unsupported arcs, especially if the infill is printed in G1
segments.

When you run M115, does it list Arcs as an option?  Another way to find out
would be to run G2 by itself.  If it reports "unknown command", then arcs
aren't supported.

fieldOfView, 2021-03-14:
If your Marlin build is compiled without support for G2/G3 arc movements, then gcode modified by this plugin will not work.
See https://github.com/MarlinFirmware/Marlin/blob/bugfix-1.1.x/Marlin/Configuration_adv.h#L857

pbezza, 2021-03-14:
That would be the issue. I didn't find much info on the plugin before installing it but I did find a couple of videos showing the improvements with it enabled which sold me. I might as well install Marlin 2.x if I'm going to upload firmware again.

closed 2021-03-14 by pbezza