fieldOfView/Cura-ArcWelderPlugin #17: Printer profile settings not registering

Asterchades, 2021-05-02:
I've created a complete custom printer profile for my use which defines a number of settings that I use globally - many of which are formulaic, which is why I don't just use the internal Cura print profile feature. Two of the options I set in this profile are "arcwelder_enable" (true) and "arcwelder_g90_influences_extruder" (also true). However, neither of these settings are "taking" when using the base printer profile, instead reverting to their default values (false for both). This is the same whether I reassign the "value" or "default_value", or both, in the printer profile.

All other options are functioning as I have specified.

The specific declarations I use are:
```
				"arcwelder_enable": { "value": true },
				"arcwelder_g90_influences_extruder": { "value": true }
```
_No trailing comma as that is the final entry in the profile._

In case it is relevant, I am using SmartAvionics' master builds of Cura, last updated April 20th (just after the 4.9 release). Plugin is v3.5.1 from the Cura Marketplace.

fieldOfView, 2021-05-02:
Specifying values in the .def.json file is going to be hard to impossible to support in this plugin, because the .def.json is loaded and parsed before the plugin adds its own settings to Cura. In doing so it replaces the settings you made.

Just in case though, could you zip up your .def.json files so I can see exactly how you are trying to do this?

Asterchades, 2021-05-02:
[Thalyn_FDM.def.zip](https://github.com/fieldOfView/Cura-ArcWelderPlugin/files/6411228/Thalyn_FDM.def.zip)

Pretty much exactly as stated. I use two profiles so I can separate out the things that I don't want across all Cura installs (like disabling Wipe for Arachne at present) or printers.

fieldOfView, 2021-05-09:
I'm considering this as "not a bug", but I'll keep the issue open as an "enhancement". I never intended the plugin values to be set in the printer definition (.def.json). You can set them in profiles (.inst.cfg) and even in xml material profiles if you wish.

Asterchades, 2021-05-09:
Sounds good to me.