fieldOfView/Cura-ArcWelderPlugin #18: Arc welder creating artifacts in Cura

PhilBaz, 2021-05-03:
I see the beauty of Arc welder but have had mixed results using it in cura. It mostly works great but has caused problems a few times.

Please see the attached images of 3 prints...the first two are sliced with different setting and using arc welder, the third has Arc welder turned off and the artifact is gone...ill attach the images as well as the G-code (github made me zip it). Is this an issue or do my settings need to change? I'm using settings very close to the defaults.

Thanks for your time and effort!

[image]
[image]
[G-code.zip](https://github.com/fieldOfView/Cura-ArcWelderPlugin/files/6415566/G-code.zip)
[image]

fieldOfView, 2021-05-03:
Is the order of the images correct? I see what looks like an underextruded horizontal line (outer wall) in the last two images, but not the first (it seems to be there still, but far less noticable). Is that the artifact I'm supposed to be looking at?

PhilBaz, 2021-05-03:
The under extrusion line is the artifact, yes... Don't know if they are in order, but the naming of the images corresponds to the G-code names. 

Thanks for the quick reply!

PhilBaz, 2021-05-03:
PS...I can run the test again, or more tests if you think it was a fluke.....Also, there was a third AW slice benchy print which ended up with the same artifact.

fieldOfView, 2021-05-03:
Let me ping @FormerLurker who created the actual [magic](https://github.com/FormerLurker/ArcWelderLib) that processes the gcode, to see if he has any ideas.

seantapscott, 2021-05-03:
Thanks for sharing.  Could be a lot of things depending on printer setup;
filament out of round, sudden draft from an opened door, untuned pid values.

Could you share filament brand and printer model, and any additional steps
you've already taken towards verifying printer tuning and filament diameter
accuracy?

PhilBaz, 2021-05-03:
Its an artillery X1 sidewinder......    Filament brand is Novofill, based on ingeo resin. I have my flow calibrated and get consistent high quality prints from the printer. 

I sliced the benchy with Arc welder 3 different ways, and one additional slice was a duplicate of one of the initial 3, but without arc welder. All three with AW exhibited the same artifact. The one without AW didn't. 

Im happy to run more tests. Any ideas?

seantapscott, 2021-05-03:
G90 checked or unchecked?  I'd also be curious what version of Marlin
you're running.  Again, just helping gather info so the troubleshooting can
be quick and effective.

PhilBaz, 2021-05-03:
G90 unchecked...Marlin FW bugfix version, March 21 2021 fw 2.0.7.2 .

No problem, i get it. And i'm happy to do more testing if you think there is something specific i should try. I appreciate your time and all the quick response.

Phil

FormerLurker, 2021-05-03:
@PhilBaz, G90/G91 influences extruder should be checked for Marlin 2.0+. Can you attach both the modified and unmodified gcode (I read in your OP that you would attach these, but can't seem to find them) please?  It may well be that having that option disabled is causing the problem, but it could also be issues with variable line width.  Are you using Cura Arachne by chance?  If so, I'm working on a patch that should fix any issues encountered there.  I had to add some checks to break apart arcs if the extrusion rate changes in the middle of an arc.  Without this, any differences in line width will be combined across the entire arc, possible causing under or over extrusion.  I should be able to tell all this from the gcode.  Feel free to [create an issue in my ArcWelderLib repository](https://github.com/FormerLurker/ArcWelderLib/issues) as well, since this probably has nothing to do with @fieldOfView's awesome plugin.

PhilBaz, 2021-05-03:
[G-code.zip](https://github.com/fieldOfView/Cura-ArcWelderPlugin/files/6416272/G-code.zip)

I'll run another test with the option checked... Files are in the zip and labled.

PhilBaz, 2021-05-04:
So the test with G90 enabled also came back with an artifact.

I sliced the benchy then save d one with AW and one without, no other changes. One difference though, now the artifact is pushing layers out rather than in. Funny, what looked  like under extrusion with G90 off looks like over extrusion now. You can also see the no AW version is pretty much flawless. And this slice is totally different than the ones pictured earlier. Ill attach pics and gcode.
[F.zip](https://github.com/fieldOfView/Cura-ArcWelderPlugin/files/6418642/F.zip)
[image]
[image]

FormerLurker, 2021-05-04:
Interesting, will look into this asap, and thank you for checking that out!

PhilBaz, 2021-05-04:
No problem...thanks for the great tools!

FormerLurker, 2021-05-09:
Update:  After an extensive review and no issues discovered with the arc welder gcode, I ran the results through the inverse processor and can clearly see the deviations!  This appears to be an issue with the arc interpolation algorithm's 'N_ARC_CORRECTION' value.  When set to 0, the artifact vanishes.  Need to look into it a bit more and figure out if there are any tweaks that can be done short of removing the small angle approximation.  In other words, this is a firmware issue.  

I have a hunch that reducing the maximum arc radius to something smaller (maybe 100 meters) would also fix the issue, but haven't tested it.

Will keep you posted.

brunoosti, 2021-09-30:
I'm getting something similar but on lines that should be straight (diagonal on the print bed, so both motors have to work together, if that matters - sidewinder x1 or anycubic i3 mega printers). Sometimes it's "stair stepping" where it should be completely straight / smooth and some other times is just one layer o anther that seems to get a little bit "out of path".
[image]
[image]
![IMG_20210901_134607](https://user-images.githubusercontent.com/70181464/135509112-82bd596d-6258-43c4-9656-59403048eb11.jpg
[image]
)

yash-fn, 2021-12-05:
Unfortunately I do not believe anyone on the Marlin side knows how the G2/G3 code works anymore and hence it's essentially unmaintained. Wouldn't count on devs doing much to fix it either since it's not really a priority issue for them. Means our little arcwelding community will have to fix it with our own effort.

FormerLurker, 2021-12-05:
Can you post the Source and welded gcode please?  Also, huge advances in m G2/G3 interpolation have been introduced in recent marlin builds.  If possible, definitely upgrade your firmware to the latest marlin build.

That being said, what version of AW are you using?  Are you maybe using cura arachne?

FormerLurker, 2021-12-05:
@brunoosti,  I didn't realize what repo this was in, doah!  Can you post to the arcwelderlib repo issues or discussions instead?  That would be the right place to investigate this upstream issue.