fieldOfView/Cura-ArcWelderPlugin #20: Artifacts or settings issue?

Dan-imal, 2021-05-21:
Small screw and nut.  One with Arc Welder Turned on and one with it off (I'm guessing you can figure out which is which).
[image]

Here's the settings.  They are what came pre-loaded.
[image]

seantapscott, 2021-05-21:
Hello there,

Always helpful to also list your printer name and firmware version.

fieldOfView, 2021-05-21:
gcode files with and without ArcWelder turned on are also welcome to analyse the difference.

Dan-imal, 2021-05-21:
Here's the GCode for my Ender 3v2
[nut & bolt 3.4m walls 20%inf 57m.zip](https://github.com/fieldOfView/Cura-ArcWelderPlugin/files/6524101/nut.bolt.3.4m.walls.20.inf.57m.zip)
Thank you.

seantapscott, 2021-05-21:
G90 Influences Extruder should be turned on for Marlin. The longer arc
commands also cause trouble with the buffer size on board the mainboard.
Need to compile from the bugfix branch of marlin while enabling certain
values.

See this thread here for more info on upgrading firmware on Creality
printers:
https://github.com/FormerLurker/ArcWelderPlugin/issues/175#issuecomment-808851235

@ldursw found the solution.

fieldOfView, 2021-07-22:
I'm closing this issue because it seems to be a combination of firmware and settings.

closed 2021-07-22 by fieldOfView