fieldOfView/Cura-ArcWelderPlugin #21: Plugin for Cura LulzBot Edition

b-morgan, 2021-06-29:
The current release of Cura LulzBot Edition 3.6.23 (CuraLE) is based on an older version of Ultimaker Cura.

I believe the plugin structure is similar but there's (obviously) no Marketplace from which to install a plugin automagically.

Would it be possible to install this plugin into CuraLE manually and if so, could you provide instructions to do so?

fieldOfView, 2021-06-29:
I have not tested this, but you may try downloading [this version](https://storage.googleapis.com/ultimaker-cura-packages/v1/packages/ArcWelderPlugin/zzHJfdU_udY4lWeMO9r9Ha-zSXKtZbLMRyRejKldEXtt/ArcWelderPlugin-v5.0.0-2021-02-06T17%3A17%3A47Z.curapackage?Expires=1940342806&GoogleAccessId=cura-packages-api-production%40stardust-193112.iam.gserviceaccount.com&Signature=gkQ0HoC%2BzKOTuhsZceazCnl9j3UTnS9L%2FCymETAsf3EbXTgJ5r3qZk2%2BH04bjwtHpbgkeJ14YU0P9kLgKWNszTdVsHd5m%2FDcmnvBINsldPBejAFN1fNJEJAvY3dLL7XktOaHPD1vmPkS7BPGRUF06E5Cz7MnenB9n1tdeX7nYRzjRRrs0yKsBIZLsCe5HPG9j0BfzQZTGckwpy04%2FfVkvYAejwRJFTvotITB2ehroCtfQiUqrLzFI1e18gTZ%2B23etM0aeATF%2BHY1MD2WMHDu3Zy0RYvj1yoZT%2FH5NJhXz65%2BBbVSIPlJ1K%2FvBWAxzFVDykSMQq3VeBUJqAU6BtjXug%3D%3D) and dropping the file onto the buildplate in CuraLE.

b-morgan, 2021-06-29:
Unfortunately, that didn't work.

[image]

seantapscott, 2021-06-29:
Was there a version of the plugin that supported Cura 3.6.2?

fieldOfView, 2021-06-29:
Then you will have to install it manually. 

* Download and extract https://github.com/fieldOfView/Cura-ArcWelderPlugin/archive/refs/heads/3.5.zip
* In Cura go to Help -> Show configuration folder.
* Enter the plugins folder, and create a folder named `ArcWelderPlugin`
* Put the files from the downloaded zip there

b-morgan, 2021-06-29:
Some progress... I got to Preferences, Plugins and Arc Welder plugin shows up, but it is not checked. I check the box and according to the note at the bottom of the window, I will need to restart the application before changes in plugins have effect. I do so but I don't have any new settings and when I return to Preferences, Plugins, the Arc Welder plugin is not checked. I have attached the .cfg and the .log file.

[3.6.zip](https://github.com/fieldOfView/Cura-ArcWelderPlugin/files/6736494/3.6.zip)

I'm guessing this is the problem:

`2021-06-29 15:10:35,471 - INFO - [(3620)-MainThread] UM.PluginRegistry.loadPlugin [222]: Plugin ArcWelderPlugin uses an incompatible API version, ignoring
`

fieldOfView, 2021-06-29:
I'm sorry, then Lulzbot have mangled Cura to a point where I have no interest in supporting it.

closed 2021-06-29 by fieldOfView

b-morgan, 2021-06-29:
Thanks for trying!