fieldOfView/Cura-ArcWelderPlugin #23: caught a bug: incorrect processing of the trajectory, risk of breaking the printer

Goodfeat, 2021-08-11:
Hello, I found a bug when processing a 3d model with a plugin from under the cura 4.10

This is the second time I've encountered this bug. This time I decided to print a gear, but something went wrong.

[image]

[image]

[image]

One of the lines extends beyond the printable part. During printing, the printer rests on the limit switch, but still tries to continue moving, causing wild noise and vibrations.

seantapscott, 2021-08-11:
What is your printer model?

Goodfeat, 2021-08-11:
Pusa, firmware repetier, but in this case, the printer has nothing to do with it. (It may also be my fault that the printer does not stop when the nozzle goes out of bounds)

here the problem is more software

seantapscott, 2021-08-11:
Yeah, as a person who's helped troubleshoot problems with ArcWelder, about
80% of them have to do with the printer's firmware settings.  ArcWelder is
creating commands that are longer, and so on some printers fills up the
command buffer faster than traditional straight-arc commands, and then we
get buffer overruns which occasionally pass along a sufficiently mangled
command that the firmware understands, but which doesn't actually make
sense, and sends it out to X400 Y300 because the command was garbled.

So yeah, any additional information on the firmware version will help.  Do
you have the output gcode file as well?

Goodfeat, 2021-08-11:
[P_24_110821_1018.zip](https://github.com/fieldOfView/Cura-ArcWelderPlugin/files/6965497/P_24_110821_1018.zip)

the firmware repetier 1.04 is custom, but the fact is that this bug already appears when the file is generated, simply by saving the file to the computer, so the printer does not in any way refer to the distortion of the file

However, I noticed that this bug disappears if you reduce the resolution accuracy of the cura file generation to 0.1mm, if you set it to 0.05 or 0.01mm. the result may not be predictable

Goodfeat, 2021-08-11:
[image]
Most likely the plugin does not have a check for the correct path creation. Perhaps, it is necessary to add functions for checking the trajectory validity, checking with the true dimensions of the printer. (Perhaps the processing of the file would take longer, but in case of an error, it would be possible to output the data to the console). It seems to me that this would help at least eliminate this defect, or at least stop the generation of the file with an error

fieldOfView, 2021-08-11:
Can you reproduce this erroneous gcode? If so, a link to a project file (File -> Save project...) would be helpful.

Goodfeat, 2021-08-11:
I cannot reproduce this bug again. I restarted the program and the bug does not appear now. it looks like this bug is floating. it takes more time to understand at what point it manifests itself

Goodfeat, 2021-08-11:
it turns out I made a small change to the model and the bug disappeared. I tried the previous version of the model and the bug showed itself again. It seems that this can manifest itself on any model and completely by accident.
[P_24_110821_1647.zip](https://github.com/fieldOfView/Cura-ArcWelderPlugin/files/6966457/P_24_110821_1647.zip)

fieldOfView, 2021-08-11:
I have not had a look at the project yet, but I'm pinging @FormerLurker who is the actual wizard.

FormerLurker, 2021-08-11:
Limit the maximum arc radius to 999 or 9999 mm (I've seen no issues with 9999, but 999 will work if you want to be extra sure) and this issue should vanish.  I've updated the default settings in the console app and the octoprint plugin (dev branches) with this fix, and it will soon be released.  It might be a good idea to change the cura plugin defaults to match (9999), and it doesn't reduce compression noticeably.  I've also added the ability to skip generation of long gcode commands as an extra safeguard in the dev branches.

Also, I analyzed your gcode, and the tools I have show the correct path:

[image]

What visualizer were you using?  It also might have an issue with long commands.  Anyway, I am 99% sure the fix (reducing the max arc radius) will solve your issue.  Also, I recommend switching dynamic precision off, and hard coding them to 3 decimals for XYZ and 5 for E.  That may limit the gcode length if the start/end gcode uses more precision than necessary.

FormerLurker, 2021-08-11:
Also @Goodfeat, regarding this:

>Most likely the plugin does not have a check for the correct path creation.

It definitely does have a ton of checks for correct path generation.  It's always possible there are bugs, I've gone to pretty extreme lengths to ensure the paths generated follow the original paths down to the provided resolution (+-0.025mm by default).  In fact, the error detection takes longer than the path generation to complete, so most of the processor time arcwelder uses is actually testing the generated paths.  I just didn't want you to think this was left out of the algorithm :)

Goodfeat, 2021-08-11:
> What visualizer were you using?
i use the repetier-host as a g-code analyzer

I double-checked the g-code file that I uploaded, it contains an erroneous line with too high values. Please pay attention to line 467333 in my g-code

they contain coordinates with the value

G3 X112.962 Y115.772 **I-161355.710 J69710.793** E2324.18811

The  repeater-host clearly shows this movement as it is (I use the repeater because other analyzers cannot handle the G2 and G3 commands)
Can your analyzer ignore this line? which analyzer are you using?

FormerLurker, 2021-08-11:
Simplify 3d.  I find it is the most accurate one when visualizing G2/g3 commands.  Also, pretty gcode viewer for octoprint is good.  I personally coded the arc move visualizer for that plugin.

FormerLurker, 2021-08-12:
The image i posted was from simplify.  I made sure that the file I checked (the first one you posted) contained g2/g3 commands.

Goodfeat, 2021-08-12:
Unfortunately I will not be able to test my code in the Simplify 3d
   The repeater-host is distributed free of charge and it displays the movements of the g-code just as well

All the same, I am haunted by the fact that in your program it is displayed differently than in the repeater host
I just look at your image and it is very different from my image along the trajectory
[image]

For some reason, your slicer shows my g-code perfectly, although when analyzing it through notepad ++, it is clearly visible that this should not be
[image]

FormerLurker, 2021-08-12:
I understand being mystified by the myriad ways in which arc commands are poorly supported.  To make up for that, I will run your gcode through marlin's arc interpolation algorithm, and will report what a different firmware will do with your gcode.  It will show each step and all movements, so we can get a basis for comparison just in case problems still remain in the core algorithm.

Goodfeat, 2021-08-12:
OK
I hope all this will help in solving the problem

P.S.
I made a comparison incorrectly in the previous post. Now showed to display the path so that there was a correct comparison
[image]

FormerLurker, 2021-08-12:
I wanted to double check my results even more, so I ran extra tests.  First, I want to make sure I am using the correct file for these tests, so here is a snapshot of line 46733 to ensure we are on the same page:

[image]

So, first I took that file, uploaded it to Octoprint and used PrettyGCode viewer to render the gcode.  This is the result:

[image]

I then ran the same file through ncviewer.com on the layer you pointed out (containing line 46733, from 45880 lines to 48910), and this is the output:

[image]

There are no errant lines visible there.  Next, I ran your original file through a port of marlin 2's arc interpolation algorithm named named ArcStraightener.exe (available in my repo), which I use for analyzing firmware issues.  Using mm-per-arc-segment=1.0 and min-arc-segments=14, I converted the G2/G3 commands from your original file to G1 commands, and this is the result in Simplify 3d:

[image]

Here are the original welded gcode file and the straightened (g2/g3 converted to g1) gcode files for review:  

[WeldedVsStraightenedGcode.zip](https://github.com/fieldOfView/Cura-ArcWelderPlugin/files/6972144/WeldedVsStraightenedGcode.zip)

I wanted to stress that I did not do this to prove anything.  Rather, I want to be absolutely certain that this algorithm is bullet proof.  I take all potential major issues reported, like this one, VERY seriously.  If there was any doubt at all that the gcode you used was incorrect, I would have spent as much time as it took to fix it.  In other words, I'm glad you posted this, and to the best of my knowledge Arc Welder produced correct gcode.  As with lots of problems I see, this is likely an issue specific to your version of repetier, and is most likely due to that single gcode line being longer than your firmware can handle.  To further put your mind at ease, I'd like to let you know that I've already implemented specific workarounds (max gcode length, altered default parameters, detailed help files explaining the issue) in both the console app and the OctoPrint plugin.  These should work to prevent problems until the firmware issues are addressed by the developer.

Would you do me a favor and post this issue on [Repeter's github](https://github.com/repetier/Repetier-Firmware) and link back to this issue?  I think this issue is best solved there, though you're welcome to download the development version of the arc welder console app if you want to use the new workarounds.  Please let me know.

Thanks!

FormerLurker, 2021-08-12:
> rehearsal firmware

I have not heard of this firmware before.  A version number and a link to the source would be appreciated.

> the problem arises. and the problem is not even in the cura, but in the post-process of the arkweider plug-in.

I believe I have provided adequate proof that the welded gcode is valid.  I don't know what more I can do.  All I can say is that the file I examined (attached in my last message) looks perfectly fine to 3 different visualizers and to marlin's arc interpolate algorithm (a port of the the algorithm that actually prints the gcode).

> The main cutting tool is cura version 4.10 and cura arachne.

Fyi, arachne support is only available in the development branch of the arcwelderlib repository.  It is still in beta.  Using the release version with arachne may result in incorrect line widths.

seantapscott, 2021-08-12:
Keeping in mind, again, Vladimir, that the "errant coordinates" you
mentioned are actually not errant at all.

Here are the I and J properties of the G2 and G3 commands:
[image: image.png]

So, the X and Y coordinates in the G3 command previously referenced would
point to where the nozzle is travelling to, the I and J point to a relative
coordinate of an arc center, and the E value refers to an absolute extruder
axis coordinate.

The best proposed solution is the original one offered to turn down Maximum
Arc Radius and disable Dynamic Precision.

Goodfeat, 2021-08-12:
> оригинальное решение, предлагающее уменьшить максимальный радиус дуги
Ok, I already did it as a temporary measure. There is a small chance that this can happen even if the radius is reduced to reasonable limits.

FormerLurker, 2021-08-12:
> and now it really becomes clear that not all editors are able to display the bug correctly

There is a second possibility: both your firmware and repetier host have the same bug interpreting arc commands.  I have seen this many times before, and have even submitted patches via pull requet to correct them.  I will take a look at the arc interpolation algorithm for your firmware and see if I can find an issue. I'd say it is likely an incorrect rotation due to a vert small angle theta betweet steps.  This is a very tricky issue for firmware and visualizers to get right.  I have had similar issues in the past.

Goodfeat, 2021-08-12:
> Есть и вторая возможность: и ваша прошивка, и хост-повторитель имеют одну и ту же ошибку

I, too, am inclined towards this. although I am confused by the very fact that out of 47,000 lines of code, only one line has large values of I in J

FormerLurker, 2021-08-12:
The bigger the radius, found by sqrt(i^2+j^2), the closer to straight the path will be.  Turning down the max radius will reduce the maximum values for I and J, but will prevent a small amount of compression.  This should prevent issues for all firmware I have seen.

Goodfeat, 2021-08-12:
I realized the microscopic arc makes the values too large. In theory, there should be a straight line, but the plugin sees a curvature and tries to process this segment as an arc.

now I think that the variable in the firmware is not capable of handling such a large value and draws the wrong line

analyzed the file z-code using PrettyGcode

[image]

in general, now I really conclude that the repetier firmware does not process this value correctly. And yes, most likely, the repetier-host analyzer does it according to the same principle as the firmware itself

Goodfeat, 2021-08-12:
I created a new issue in the branch https://github.com/repetier/Repetier-Firmware/issues/1081#issue-967910156
I really hope that the developer @repetier will be able to help in resolving the situation

FormerLurker, 2021-08-13:
@Goodfeat, can this issue be closed?

Goodfeat, 2021-08-13:
Hello,
So far, there has been no response from the developer to clarify the situation.
However, there are already several solutions to this problem in this thread.
And I think yes, you can close the topic.
thank you

repetier, 2021-08-13:
Can you give an example of a valid arc that is drown incorrectly? Only thing is you can not do 360° arcs as they are indifferent and nearly 360° are numerically instable due to float resolution.

Goodfeat, 2021-08-13:
Hello @repetier 
Unfortunately, I can only give an example of what is obtained in the g code after processing by the plugin,
the slicer cuts the part with a resolution of 0.01mm,
most likely the bug occurs as a result of incorrect processing of the command of a small arc.

And judging from the discussion, other renderers do not see the problem in this section of the r-code, while the repetier-host shows an obvious problem. the firmware-repetier also tries to execute this command incorrectly

[image]

repetier, 2021-08-13:
Ok I checked it in host. It is a numerical problem that comes from the test if a circle is closed. This causes non circles like this to get converted into 360° circles with radius around 180m. I have now reduced this to a simple line sine a circle of 0.000006rad can be approximated very good into one line segment.

At least repetier firmware has the same code. But I have to think about it. I'm quite sure this breaks with 360° circles when they are rounded to 0.00001rad angle from numeric instabilities. It is a problem that you can only have tine circle moves or 360° circles correct all the time, but not both.

On the other side I think no generator should create any of both anyway. 360° should be 2 180° segments and so tiny lines really should be kept as lines. It is just forcing unnecessary problems. Maybe I add for identical target position a 360° special case.

Goodfeat, 2021-08-13:
Ok, it would be a good idea to fix this in both the firmware and the host. The printer, due to this error, may ignore the printable area limits and begin to break.

this has happened on my printer twice already. The printer simply reached the end of the printable area and hit the rack, but continued to move.

thank you

repetier, 2021-08-13:
I think fixing on arc welder would also a good idea:-) After all it is generating this edge cases that force all other software to limits and users might be on older versions if I fix it in host/firmware. Plus I think there will still be undetectable cases.
I'm just thinking that such a big radius is already a sign of such a case. Thinking about it it can not be 360° from the size.

Goodfeat, 2021-08-13:
`and users might be on older versions if I fix it in host/firmwaree`

I can suggest for the host repeater to make a separate option in the settings for corrected firmwares.
This option is required to test that the firmware does not repeat error the repetier-host .

repetier, 2021-08-13:
Too complicated. I will try to add a fix for our software to handle it correctly of course, but I can not check if any firmware can handle it correctly. Especially not firmwares I don't control.

Goodfeat, 2021-08-13:
I understand you, then I can use the old and new versions of the repetier-host for tests.
I have a firmware repetier on two printers, most likely I will need to manually correct the firmware since the firmwares has been customized

repetier, 2021-08-13:
Yes, just watch for update on 1.0.5 dev. Not sure when I get the time to do that. Currently too much to do.

Goodfeat, 2021-08-13:
Ok, I'll keep an eye on the new firmware release.
also in branch [https://github.com/repetier/Repetier-Firmware/issues/1081#issue-967910156](url) was open a topic on this issue. As soon as the fix appears, the topic can be closed.

fieldOfView, 2021-08-13:
I'm happy my plugin played a minute role in improving Repetier Firmware.

The next release of this plugin will ship with a smaller default setting for the Maximum Arc Radius setting (as per the ArcWelder console utility), further decreasing the likeliness of this happening.

closed 2021-08-13 by fieldOfView

seantapscott, 2021-08-13:
Well, he also had dynamic precision on, and I think that's already off by
default.  Its possible that turning that on too may have caused this.

FormerLurker, 2021-08-13:
@seantapscott, in this case I don't think the dynamic precision was an issue.  I added that parameter at the request of the Smoothieware devs to preserve the full precision of the source commands, which was apparently helpful for testing purposes, but it has caused some pain related to the buffer overrun problems you are so familiar with :)

Also, fyi, I will be adding all future comments to the [new issue within the firmware repo](https://github.com/repetier/Repetier-Firmware/issues/1081).  I'm now working on adding the current Repetier arc interpolation algorithm to ArcStraightener to help with any similar issues in the future.

Thank you all!

Goodfeat, 2021-08-13:
> Well, he also had dynamic precision on, and I think that's already off by default

Hello @Seantapscott. This bug appeared even before the introduction of the "dynamic precision" option on another model. Maybe a couple of months ago.

I just looked for the reason in the settings, reduced the accuracy of the arcs and reduced the resolution of the model slicing.
Then, as it seemed to me, I achieved so that this bug did not appear anymore.

But this bug appeared again, and on a different model. So dynamic accuracy in this case has nothing to do with it, I just turned on additional options in the hope that the glitch would disappear.

Thanks to the developers for their help in solving this problem.

Thank you all!