fieldOfView/Cura-ArcWelderPlugin #24: ArcWelder default Maximum Arc Radius too large for 8-bit print controllers

Risto-H, 2021-08-30:
This parameter is set to 1000000.0 by default and it sometimes causes a situation where the I or J coordinates of G2 or G3 commands can become very large. This seems to be a problem for at least Pusa I3 family of printers that use Arduino based Einsy Rambo print control boards.

As an example, here's a code line from Planeprint Notos model fuselage6 part after slicing with Cura 4.10.0 and ArcWelder with defaults:

G1 X118.19 Y110.173 E3981.46029
**G3 X118.683 Y75.091 I664677.688 J9323.032 E3982.30632**
G1 X118.141 Y67.282 E3982.49508

Note the I coordinate that is quite large. This produces a clearly visible print error.

I changed the Maximum Arc Radius to 10000.0 and this seems to have solved the issue. However, I do not know how large radii the Prusa I3 can handle but I'm trying to find out from their forum.

Here's the same code snippet after changing the setting:

G1 X118.19 Y110.173 E3981.46029
**G3 X118.398 Y95.132 I7227.325 J92.424 E3981.82301
G1 X118.683 Y75.091 E3982.30632**
G1 X118.141 Y67.282 E3982.49508

Note that after the change, the problematic line was split into two parts and the G3 part has much smaller radius.

This is not really a bug but a request for improvement that would help users with 8-bit printers. An image showing the print error is attached.

[image]

fieldOfView, 2021-08-30:
The next release of the plugin will come with a new version of the ArcWelder command line tool, and I will change the default value to reflect the default value of the ArcWelder command line tool.

Risto-H, 2021-09-02:
I confirmed that the suggested change works by printing the part again with new settings. The print result was perfect which means that for Prusa I3 series printers Maximum Arc Radius of 10000.0 is a good choice.

In the same time it looks like no one knows what the maximum radius really is that Prusa I3 can print without clear inaccuracy. Perhaps we don't even need to know since setting it to 10000.0 eliminates visible problems and avoids performance issues due to large numbers of straight sections in an arc shaped print section.

brunoosti, 2021-09-30:
I'm getting something similar but on lines that should be straight (diagonal on the print bed, so both motors have to work together, if that matters - sidewinder x1 or anycubic i3 mega printers). Sometimes it's "stair stepping" where it should be completely straight / smooth and some other times is just one layer o anther that seems to get a little bit "out of path". Maybe something to do with max radius as the plugin miss interprets a straight line with a curve?
[image]
[image]
[image]
[image]