fieldOfView/Cura-ArcWelderPlugin #25: Arc Welder Cura extension causing badly placed skirt and odd movements at higher layers

michaelmadden, 2021-12-13:
Hello,

I've been successfully using Arc Welder with Cura for quite a while now, but in the last few days, I've seen some real interesting behavior.

I was originally on Cura version 8.10.x, and noticed that the prints I sliced had the skirts placed off kilter to the actual object I was printing. I also noticed that at some of the higher layers, there were odd movements that did not make sense given the shape of the object.

I re-flashed the firmware on my machine (Ender 3 Pro) thinking that was the issue, to no avail. I played with various settings, checked belts, tightened things, etc... with the same result.

I then decided to update Cura to version 8.12.1, and wound up having the same problems, despite many settings changes in an effort to resolve the issue.

I uploaded the Gcode to an online Gcode viewer and confirmed that it was the Gcode and not my printer.

On a whim, I decided to disable Arc Welder, re-slice and re-upload to Gcode viewer. The new Gcode looks flawless.

I'm not sure what happened, but it appears Arc Welder is the culprit, as I'm now printing the most recent Gcode and it is laying down just fine.

I've attached the STL I was using, as well as the two Gcode files, clearly labeled as to what they are.
NOTE: it appears GitHub doesn't allow upload of .stl or .gcode extensions. I've zipped the three files together.

[arcWelder issue.zip](https://github.com/fieldOfView/Cura-ArcWelderPlugin/files/7704743/arcWelder.issue.zip)

fieldOfView, 2021-12-13:
By "off-kilter", do you mean like this?
[image]

If it prints like this, then your printer firmware is not configured to print arcs. If your firmware does not support arcs, you should not use ArcWelder, because it introduces arcs, and your printer will just skip them.

michaelmadden, 2021-12-13:
Precisely. It appears I was a bit off on my timeline of events. By re-compiling and re-flashing my firmware, I must have turned support for Arcs off.

Many apologies for the confusion, and thank you very much for the quick help!

closed 2021-12-13 by michaelmadden