fieldOfView/Cura-ArcWelderPlugin #26: PlugIn "Arc Welder" will not start / load after update Cura 4.13.0b -> 4.13.0

SpRiNgToIfEl, 2022-01-13:
1. Execute "Ultimaker_Cura-4.13.0-amd64.exe", check "deinstall old version" / "keep settings"
2. Start Cura, the PlugIn will not start / loaded.
3. activate the check-button, restart cura, same issue
4. deinstall PlugIn, reinstall PlugIn
5. Same issue as above
[image]
[image]

fieldOfView, 2022-01-13:
Please add the Cura.log log file. which can be found via Help -> Show configuration folder. Without it I cannot see why this is happening.

fieldOfView, 2022-01-13:
Thanks for the logs. I am looking into them now.

fieldOfView, 2022-01-13:
It looks like there may be a dependency (dll) for the ArcWelder executable missing in your system.

Could you download the application "Dependencies" from https://lucasg.github.io/Dependencies/, extract that zip and run `DependenciesGui.exe`?

In the Cura configuration folder, go to `plugins\ArcWelderPlugin\ArcWelderPlugin\bin\win64` and drag and drop `ArcWelder.exe` into the DependencesGui window. Please post a screenshot of the window. That will tell us what is missing.

SpRiNgToIfEl, 2022-01-13:
Attached the screenshot of DependencesGui

[image]

fieldOfView, 2022-01-13:
In a previous version of the ArcWelder.exe build, the dependency was baked in ("statically linked"). As a quick workaround, download and extract [vcruntime140_1.zip](https://github.com/fieldOfView/Cura-ArcWelderPlugin/files/7863652/vcruntime140_1.zip), and put the dll in the same folder as ArcWelder.exe. Note that generally I would advice against downloading dlls from places. Alternatively, you may want to try installing the visual c runtime from here: https://docs.microsoft.com/en-US/cpp/windows/latest-supported-vc-redist

blazewicz, 2022-01-13:
Any advice on how to fix this on other platforms? I have same issue on macOS.

blazewicz, 2022-01-13:
Or maybe I'm facing a different issue with similar symptoms. I've checked 4.12.1 and ArcWelder plugin is also grayed-out. I'm running macOS on Apple Silicon chip, I guess binary isn't compatible with `arm64` arch.

fieldOfView, 2022-01-13:
Probably a different issue with the same symptom. But it al starts with your cura.log.

SpRiNgToIfEl, 2022-01-13:
> Or maybe I'm facing a different issue with similar symptoms. I've checked 4.12.1 and ArcWelder plugin is also grayed-out. I'm running macOS on Apple Silicon chip, I guess binary isn't compatible with `arm64` arch.

> In a previous version of the ArcWelder.exe build, the dependency was baked in ("statically linked"). As a quick workaround, download and extract [...]

i have download the dll and copy it in the folder. arcwelder works now fine with cura 4.13.0.
thanks for your help!

why is the dll missing? Arcwelder worked great with cura 4.13.0b...

fieldOfView, 2022-01-14:
@blazewicz, I cannot help you without logs...

closed 2022-05-16 by fieldOfView