fieldOfView/Cura-ArcWelderPlugin #27: Request: Next SDK Major Version

edm00se, 2022-05-16:
## Question

What's Needed to Bump the SDK to the Next Major Version?

## Why

My limited check shows it may just need a string added to the sdk array in `plugin.json`. This is to support the AW plugin working in Cura 5.0(b1) which I've been doing some limited but successful testing with so far.

## What's Going On

I ask as my testing of Cura 5.0b1 shows that the plugin doesn't show in the marketplace or load on launch. Evidently Cura 5.0 will require a major version change of the SDK, so armed with that knowledge I found the cura application dir, then the ArcWelder plugin dir, and added `"8.0.0"` to the array of `supported_sdk_versions` in the `plugin.json` like a ham fisted barbarian. This enabled the plugin in Cura 5.0b1 for me, as I already had the plugin locally installed before going up to 5.0b1 (3.5.2 plugin version) and I did the following for comparison:

- sliced CHEP cube without AW enabled in cura w/ my usual profile, it generated a 458Kb file
- sliced the same CHEP cube STL with AW enabled in cura with no other profile changes, which generated a 426Kb file
- processed a copy of the first non-enabled gcode through the CLI tool (v1.2.0) and got a 424Kb file

Loading a preview of the generated gcode with AW both via plugin and CLI seem consistent with each other. Separately there may be an issue with cura displaying my AW processed gcode (the skirt looks funky, but seems arc path related); but they were consistent with each other.

I'm going to run a print of my plugin generated gcode after the manual local change to see how it turns out, although with my experience of Cura 5.0b1 so far I'd be surprised if it wasn't fine.

fieldOfView, 2022-05-16:
A prerelease and instructions for a new version of this plugin are available here:
https://community.ultimaker.com/topic/40313-fieldofview-plugins-for-cura-50/

edm00se, 2022-05-16:
Excellent. I'm not sure how I missed that but seems that it's tracking. Thanks!

fieldOfView, 2022-05-16:
I'll leave this open for others to find, until the new version is available in the Marketplace.