fieldOfView/Cura-ArcWelderPlugin #28: Klipper question - Firmware Compensation

derDoc81, 2022-05-25:
The Firmware Compensation applies to Klipper but can be fixed with changing settings.
My problem: I don't find any information what would be the best setting for Klipper any what should be changed! :D

fieldOfView, 2022-05-27:
My plugin is just the "glue" between Cura and the Arc Welder algorithm. I have basic knowledge of what the ArcWelderlib does, but for indepth questions about the settings for specific firmware flavors, please see https://github.com/FormerLurker/ArcWelderLib

fieldOfView, 2022-05-27:
From the ArcWelderlib readme:

> If you are running (...) Klipper, you might consider enabling Firmware Compensation. This is dependent on your firmware settings, but in general you will want to set MM Per Arc Segment to 1.0 and Min Arc Segments to 14

closed 2022-11-09 by fieldOfView