fieldOfView/Cura-ArcWelderPlugin #29: Installed on version 5.1 of Cura on MacOS 12.5.1 but nothing happens to gcode

deepcoder, 2022-09-05:
I'm not clear on the necessary steps to invoke ArcWelder during a slice operation. I installed the ArcWelder plugin from Cura. ArcWelder's options appear in the preferences. It is not clear from your documentation whether you need to install the ArcWelder library in addition to the plugin. Any also it is not clear how to get ArcWelder to run on gcode from within Cura.

Thank you for your work and help.

```
2022-09-04 17:48:36,601 - DEBUG - [MainThread] ArcWelderPlugin.ArcWelderPlugin.__init__ [77]: Using ArcWelder 1.2.0
2022-09-04 17:48:36,601 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [493]: Loaded plugin ArcWelderPlugin 3.6.0
```

fieldOfView, 2022-09-06:
I'll need a full log to diagnose what is going on.

>  it is not clear how to get ArcWelder to run on gcode from within Cura.

See the readme (most notably the first item in the "Arc Welder Settings" section: https://github.com/fieldOfView/Cura-ArcWelderPlugin#readme

fieldOfView, 2022-11-09:
I cannot help you without logs. I can only assume that you forgot to enable Arc Welder, in which case your question is answered above.

closed 2022-11-09 by fieldOfView