fieldOfView/Cura-ArcWelderPlugin #30: Extruder goes bezirk

j-snijder, 2022-11-09:
Hi.

On the ender 3pro with touchscreen there is a different firmware then the stock marlin fw on the ender 3 models., this having said the following occurs.

When I install arc-welder and run it with the standard settings (so nothing is set to on) the extruder goes bezirk and when printing is pushes a big amount of filament through the nozzle up to the point the extruder is slipping and grinding dangerously. When you do not stop this it will distroy your filament and your extruder motor because of this.

what can be done to get a normal extrusion speed, as set in the slicer settings.

This happens at Cura 4.8.n, Cura 5.1.n and Cura 5.2.n and also in Octo4a so the  problem clearly is in the slicer g-code generation after arc-welder is installed.

No I've reinstalled it and arcs are really bad looking and ugly printed, it seems something ore some settings are left behind after uninstall ?

fieldOfView, 2022-11-09:
This plugin is just the glue between Cura and the ArcWelder. For support for ArcWelder itself, see https://github.com/FormerLurker/ArcWelderLib

> it seems something ore some settings are left behind after uninstall ?

Once the plugin has been uninstalled, it can no longer affect your gcode. ArcWelded gcode does not make changes on your printer. If you have uninstalled the plugin and prints are looking bad, then ArcWelder nor this plugin is at fault.

j-snijder, 2022-11-09:
Okay, thanks for your fast reply :)