fieldOfView/Cura-ArcWelderPlugin #31: "Out of Order" for Arcs, multiple artifacts and unnecissary moves

Snoman002, 2022-12-13:
Cura 5.2.1
ArcWelder 3.6
Ender 3 v1 8bit

First tests with ArcWelder and noticed that it creates a rather disordered printing pattern on circles/circular  out of the slicer.  In short on a flat round sufrace printed with concentric top and bottom pattern the arc seem placed at random.  Ideal situation would be consecutive arcs working inward or outward, in this case it appears to randomly place the arcs resulting in the printer jumping from an arc near the middle, to an arc near the edge, to the very center, then back to the middle, then outside, then center, etc.
Image of the cura playback approximately half way through the first layer.

[image]

This obviously results in a very ugly layer (partially complete in this picture)
![image](https://user-images.git

[Arc Test.zip](https://github.com/fieldOfView/Cura-ArcWelderPlugin/files/10222923/Arc.Test.zip)
hubusercontent.com/62316078/207467941-96931e90-9a23-452e-a50d-6ef10c610b5d.png)

seantapscott, 2022-12-14:
So you sliced it and then took the arcwelder-processed gcode and put it
into Cura to view the path?

Snoman002, 2022-12-14:
This was the preview  in Cura while using the Cura ArcWelder plugin.

But...
I had not used concentric top and bottom layers in the newer version of Cura.  Previous use of concentric had it printing in order (outside in/inside out) as far as I remember thus I assumed the issue was in arc welder.  Testing cura now and it puts it out of order and ArcWelder is just following cura.
Frankly, I didn't even think that this was an issue out of Cura!

Cura with ArcWelder uninstalled
[image]

seantapscott, 2022-12-14:
Arcwelder doesn't do anything until you slice and then output.

Snoman002, 2022-12-14:
uh, ok, sure.  thanks

fieldOfView, 2022-12-14:
In addition to what @seantapscott said, the gcode viewer in Cura does not display curves. It just skips them in the viewer.

closed 2022-12-14 by fieldOfView