fieldOfView/Cura-ArcWelderPlugin #36: Cura 5.5 engine plugins

psuchris, 2023-11-10:
Are there any plans to put the functionality of this plugin into a new engine plugin? Towards the bottom of [this page](https://ultimaker.com/learn/ultimaker-cura-5-5-stable-release-notes/) it explains them.

This would at the very least allow a more accurate print time estimate, as currently the arc optimizations are applied on saving the file if I understand correctly. I know for round prints arcs can save a bunch of time but it's not obvious how much faster a print is with the current plugin's limitations. It hopefully would also allow the arc movements to show up in the preview as well.

fieldOfView, 2023-11-19:
> This would at the very least allow a more accurate print time estimate,

How so? The print time is calculated in CuraEngine, and as far as I know the estimator in CuraEngine does not understand arc movements. The change in toolpaths from multiple linear to arc moves should not affect print time much. The only thing that would affect print time is prevention of stutters in many subsequent linear moves, and those are not calculated in CuraEngine in the first place.

>  It hopefully would also allow the arc movements to show up in the preview as well.

It wouldn't make Cura magically understand arc gcode commands. Adding them would still require a lot of work in Cura (vs CuraEngine).

Sure, it could be nice to completely rewrite the plugin, parts of Arc Welder itself, and parts of Cura, but I currently do not see a huge benefit.

chrisnuzum, 2023-11-19:
I understand now, sorry for the confusion. I must have seen an older video about print time speed-ups with 8-bit boards because of stuttering from all the individual line commands. But yes, that wouldn't show up in the Cura calculation.

I agree that it doesn't make sense to rewrite it, I'll close the issue.