fieldOfView/Cura-ArcWelderPlugin #40: Arc welder giving error code

Heymap, 2024-06-18:
I was trying to print a benchy with arc welder on and tried to save it to the SD card and I gives me an error. And when I turn it off it save normally and prints the part.
[image]

fieldOfView, 2024-06-18:
*with* a cura.log, please...

Cura.log can be found via Help -> Show configuration folder.

Heymap, 2024-06-18:
> _with_ a cura.log, please...
> 
> Cura.log can be found via Help -> Show configuration folder.

I'm confused srry someone told me to post this on GitHub from reddit. I'm new to github

fieldOfView, 2024-06-18:
Hi, /u/ahoeben here.

DorianVasco, 2024-09-04:
A recent cura.log looks like this (using Cura 5.8.1):
`2024-09-04 21:49:10,060 - ERROR - [MainThread] UM.Qt.Bindings.OutputDeviceManagerProxy._writeToDevice [163]: Exception: Unable to write to file EN4PRO_Seal: unsupported operand type(s) for /: 'NoneType' and 'int'
2024-09-04 21:49:10,061 - ERROR - [MainThread] UM.Qt.Bindings.OutputDeviceManagerProxy._writeToDevice [163]: Traceback (most recent call last):
2024-09-04 21:49:10,062 - ERROR - [MainThread] UM.Qt.Bindings.OutputDeviceManagerProxy._writeToDevice [163]:   File "UM\Qt\Bindings\OutputDeviceManagerProxy.py", line 150, in _writeToDevice
2024-09-04 21:49:10,062 - ERROR - [MainThread] UM.Qt.Bindings.OutputDeviceManagerProxy._writeToDevice [163]:     device.requestWrite(nodes, file_name, limit_mimetypes, file_handler, **kwargs)
2024-09-04 21:49:10,063 - ERROR - [MainThread] UM.Qt.Bindings.OutputDeviceManagerProxy._writeToDevice [163]:   File "C:\Users\mail\AppData\Roaming\cura\5.8\plugins\MoonrakerConnection\MoonrakerConnection\MoonrakerOutputDevice.py", line 86, in requestWrite
2024-09-04 21:49:10,063 - ERROR - [MainThread] UM.Qt.Bindings.OutputDeviceManagerProxy._writeToDevice [163]:     self.writeStarted.emit(self)
2024-09-04 21:49:10,064 - ERROR - [MainThread] UM.Qt.Bindings.OutputDeviceManagerProxy._writeToDevice [163]:   File "UM\Signal.py", line 216, in emit
2024-09-04 21:49:10,064 - ERROR - [MainThread] UM.Qt.Bindings.OutputDeviceManagerProxy._writeToDevice [163]:     self.__handleEmitIndirect(*args, **kwargs)
2024-09-04 21:49:10,065 - ERROR - [MainThread] UM.Qt.Bindings.OutputDeviceManagerProxy._writeToDevice [163]:   File "UM\Signal.py", line 323, in __handleEmitIndirect
2024-09-04 21:49:10,065 - ERROR - [MainThread] UM.Qt.Bindings.OutputDeviceManagerProxy._writeToDevice [163]:     self.__performEmit(*args, **kwargs)
2024-09-04 21:49:10,066 - ERROR - [MainThread] UM.Qt.Bindings.OutputDeviceManagerProxy._writeToDevice [163]:   File "UM\Signal.py", line 361, in __performEmit
2024-09-04 21:49:10,066 - ERROR - [MainThread] UM.Qt.Bindings.OutputDeviceManagerProxy._writeToDevice [163]:     signal.emit(*args, **kwargs)
2024-09-04 21:49:10,067 - ERROR - [MainThread] UM.Qt.Bindings.OutputDeviceManagerProxy._writeToDevice [163]:   File "UM\Signal.py", line 216, in emit
2024-09-04 21:49:10,068 - ERROR - [MainThread] UM.Qt.Bindings.OutputDeviceManagerProxy._writeToDevice [163]:     self.__handleEmitIndirect(*args, **kwargs)
2024-09-04 21:49:10,068 - ERROR - [MainThread] UM.Qt.Bindings.OutputDeviceManagerProxy._writeToDevice [163]:   File "UM\Signal.py", line 323, in __handleEmitIndirect
2024-09-04 21:49:10,069 - ERROR - [MainThread] UM.Qt.Bindings.OutputDeviceManagerProxy._writeToDevice [163]:     self.__performEmit(*args, **kwargs)
2024-09-04 21:49:10,069 - ERROR - [MainThread] UM.Qt.Bindings.OutputDeviceManagerProxy._writeToDevice [163]:   File "UM\Signal.py", line 357, in __performEmit
2024-09-04 21:49:10,069 - ERROR - [MainThread] UM.Qt.Bindings.OutputDeviceManagerProxy._writeToDevice [163]:     func(dest, *args, **kwargs)
2024-09-04 21:49:10,070 - ERROR - [MainThread] UM.Qt.Bindings.OutputDeviceManagerProxy._writeToDevice [163]:   File "C:\Users\mail\AppData\Roaming\cura\5.8\plugins\ArcWelderPlugin\ArcWelderPlugin\ArcWelderPlugin.py", line 193, in _filterGcode
2024-09-04 21:49:10,070 - ERROR - [MainThread] UM.Qt.Bindings.OutputDeviceManagerProxy._writeToDevice [163]:     global_container_stack.getProperty("arcwelder_tolerance", "value") / 100
2024-09-04 21:49:10,070 - ERROR - [MainThread] UM.Qt.Bindings.OutputDeviceManagerProxy._writeToDevice [163]: TypeError: unsupported operand type(s) for /: 'NoneType' and 'int'
2024-09-04 21:49:10,531 - INFO - [MainThread] SliceInfoPlugin.SliceInfo._onRequestFinished [316]: SliceInfo sent successfully
`
Happens when using 'send to printer' as well as 'save file as'

DorianVasco, 2024-09-04:
Well, it seems to directly relate to https://github.com/fieldOfView/Cura-StartOptimiser/issues/9